michaelkamprath/ShiftRegisterLEDMatrixLib #2: DM13A and single color LED

julienduhavre, 2018-02-22:
Hi ! 
Nice project here !

I have a problem, I don't know how to control a panel with 20x20 leds using DM13A chip.
This panel is one color LED (green).

How to create an image and write on the panel ?
And how to create animation ? Because the lib is "RGBLedAnimation" and no "LedAnimation".

I think more documentation are welcome 😄 

Thank you very much !

michaelkamprath, 2018-02-23:
Do you have a circuit diagram that shows how you have wired the matrix and DM13A? That would be helpful in me advising how to configure a `LEDMatrix` object. 

You are right, `RGBLEDAnimation` was not written generically and thus doesn't work with the `LEDMatrix` single-color matrix controller. This library would still be a "work in progress". Pull requests are welcome!

julienduhavre, 2018-02-23:
Hi ! Thank you for your answer.
I only have a picture of my panel

https://imgur.com/a/zVoqX

michaelkamprath, 2018-02-25:
I would really need to see the circuit diagram to give you advice on whether this library will even support the matrix you have. Most commercial matrices like that require a different driver approach as they don't use a shift register for the row power switching. Furthermore, despite your claim of the matrix using a DM13A chip, I don't see one in the picture (then again, the picture doesn't have good resolution so I can't read the chip marking, but I don't see a 24P DIP that the DM13A is.

julienduhavre, 2018-02-26:
I don't have unfortunally :disappointed: 
But I'm sure it's a DM13A chip behind the panel (another picture I taked while I removed the protection)
You can see on right corner (not high quality but discernable) the DM13A name written on.

On original card, there is a 74HC245D to convert 3.3 to 5V (I guess) : 

https://imgur.com/gallery/f0Dks

julienduhavre, 2018-03-01:
So, what do you think :smile: ?

michaelkamprath, 2018-03-01:
That does look like a DM13A. But again, not knowing the actual circuit prevents me from giving you any advice on how to use this library. There are many ways the DM13A could be used that would not be supported by this library.

michaelkamprath, 2018-04-29:
@julienduhavre Given the lack of follow up, I am going to close this issue. Please feel free to re-open it if you are able to give me details on the circuit you are trying to us this library to drive.

closed 2018-04-29 by michaelkamprath