michaelkamprath/ShiftRegisterLEDMatrixLib #4: don't know how to include image

tarbear123, 2018-11-30:
i have it coded but i don't know how to include image 
#include <Arduino.h>
#include "TimerAction.h"
#include "LEDMatrixBits.h"
#include "SPIConnection.h"
unsigned int rows, 32
unsigned int columns,32
bool columnControlBitOn = LOW,
bool rowControlBitOn = LOW,
unsigned int interFrameOffTimeMicros = 0,

#if defined ( ESP8266 )
      int slavePin = D8 
#else
      int slavePin = 10 
#endif

LEDMatrix();
  
  /**
   * Should be called before any operations against this object is performed.
   */
 void setup();
  
  /**
   * Returns the image buffer for this matrix object. This is the 
   * image buffer that drawing should be done to.
   *
   * @return a MutableGlyph object reference that is the matrix's image buffer.
   */ 
  MutableGlyph& image(void)       { return *_screen_data; }

  /**
   * Returns a const reference to the image buffer for this matrix object. This is the 
   * image buffer that drawing should be done to, but is const here and thus should
   * be only read from.
   *
   * @return a const MutableGlyph object reference that is the matrix's image buffer.
   */ 
#endif //__SCREEN_H__

michaelkamprath, 2018-12-01:
I'm not sure what you are asking here. The code you post seems to be bits copied from various aspects of the library and then combined with a `setup()` call. What are you trying to accomplish? Have you reviewed the examples provided for how to use the library?

tarbear123, 2018-12-02:
i'm trying to accomplish displaying gifs on a 32x32 matrix

michaelkamprath, 2018-12-03:
First, you need to ensure ensure that you 32x32 matrix has the circuit that this library supports. Specifically, this library only works with matrices that use shift registers for row and column control. Most 32x32 matrices you can buy at Adafruit or on eBay are a different circuit that this library was not designed to be used on.

Second, there is no provisions in this library as-is for loading GIF images. You can certainly write one if you desire, and pull requests are welcome. But sadly, more code is needed to display a GIF image on a LED matrix that this library controls.

michaelkamprath, 2018-12-06:
@tarbear123 If you have no further questions, I will be closing this issue.

michaelkamprath, 2018-12-10:
Closing due to inactivity. Please reopen if needed.

closed 2018-12-10 by michaelkamprath