michaelkamprath/ShiftRegisterLEDMatrixLib #5: ESP32 adapt

gurvansz, 2018-12-17:
Dear,
Could you help me toadapt your lib for an ESP32 board?
Lot of error messages when building the code.
Thanks a lot

michaelkamprath, 2018-12-19:
@gurvansz  The code already has been verified to work with a NodeMCU and a Wemo. Specifically what device are you trying to run it on?

michaelkamprath, 2018-12-22:
It turns out I had an ESP32 board in may inventory, and I found some documentation on how to create interrupt drivers on ESP32. I'll work on an update to support ESP32.

michaelkamprath, 2018-12-22:
@gurvansz I was able to create a branch on the library that now supports ESP32. The [branch is `esp32`](https://github.com/michaelkamprath/ShiftRegisterLEDMatrixLib/tree/esp32). Please check it out and tell me if it works for you.