michaelkamprath/ShiftRegisterLEDMatrixLib #9: Schematic and code still in sync?

sixtyfive, 2019-03-15:
So with the wiring from #8,  the uC is talking fine to the matrix, hardware-wise. Just to be sure I also looked at the SPI data on the scope and that seems fine, too, as far as expectations go. Matrix wiring is the same as the schematic (yes, I did check multiple times and also had another person check). The 74HC595 are TI and [are specified]http://www.ti.com/lit/ds/symlink/sn74hc595.pdf) for VIH = 3.15V @ VCC = 4.5V, so my VIH = 3.3V @ VCC = 5V are well within range.

Despite all of that, the LEDs are behaving in a weird way: 

* `pixel(0,0) = RED_COLOR` => D1 glows red
* `pixel(0,0) = WHITE_COLOR` => D1, C1, B1 all glow red
* `pixel(0,1) = RED_COLOR` => A1 glows red
* `pixel(0,0) = RED_COLOR; pixel(0,1) = RED_COLOR;` => D1 and A1 glow red
* `pixel(0,0) = WHITE_COLOR; pixel(0,1) = WHITE_COLOR;` => D1, C1, B1, A1 glow red and D4, C4, B4, A4 glow green and blue

My setup is:

[image]

```
RGBLEDMatrix leds(4, /* rows, which are expected to have shared power to the LEDs */
                  4, /* cols, which are expected to be powered separately */
                  RGBLEDMatrix::INDIVIDUAL_LEDS, /* one of INDIVIDUAL_LEDS, RGB_GROUPS or RBG_GROUPS */
                  LOW, /* HIGH or LOW to turn on a column? */
                  LOW, /* HIGH or LOW to turn on a row? common anodes + PNP -> "LOW" */
                  0, /* off-time between frames */
                  D5 /* latch pin (=5) */);
```
Any idea what might be going wrong here?

michaelkamprath, 2019-03-16:
@sixtyfive Your wiring for the columns isn't what this library would expect. Please take a closer look at the [README section on bit layout](https://github.com/michaelkamprath/ShiftRegisterLEDMatrixLib#bit-layouts) and compare the library's bit layout with yours. Note that in a 74HC595, bit 8 (Q7 in the 74HC595) is the MSB bit and bit 1 is the LSB, so bit 8 of the second 74HC595 should go to the left most Red line of column 1, and bit 1 of the first (right) 74HC595 should got to the transistor's base for row A. I believe you are using a old circuit design of mine from a separate project ([here](https://github.com/michaelkamprath/rgb-led-matrix/blob/master/hardware/4x4-rgd-led-matrix.fzz)). This library has evolved past that 4x4 design. The 8x8 and 10x10 designs in the same repository are compatible with this library.

sixtyfive, 2019-03-16:
Indeed, that's the design I'm using. I was under the impression that they're all the same, just different sizes (my assumption, my bad). Back to the soldering iron then. Thank you for the explanation!

closed 2019-03-16 by sixtyfive