michaelkamprath/ShiftRegisterLEDMatrixLib #15: Colores individuales

alomdj00, 2021-01-06:
Hola, gracias por compartir esta gran libreria.

Estoy tratando de hacer una matriz de 16 rows por 4 cols,  de esta forma utilizar 28 pins de los hc595, ademas por facilidad del circuitos esta seria para mi la mejor opcion, sin embargo no tengo fritzing y por problemas en el pc no me deja instalarlo, por lo que no he podido ver los esquematicos  que tienes en el otro repositorio. 

Por lo pronto he estado haciendo una simple matriz de 2x2, y he subido el plasma de 8x8 cambiando simplemente la definicion  2x2 y esta funcionando, sin embargo cuando subo el rgb colors 4x4 cambiado a 2x2, algunos leds no encienden. Me he basado en una imagen que hay en otra pregunta ya cerrada anteriormente, basicmente me ha quedado asi. De Q0 rojo, Q1 azul, Q2 verde, Q3 rojo, Q4 azul, Q5 verde, Q6 y Q7 van a las rows, esta tal cual la imagen solo que de 2x2, y algunos leds no encienden.
Si pudieras postear una imagen del esquematico de una 8x8 o la que me podria servir para basarme a especie de guia para  contruir la de 16rows por 4cols,

Por otra parte no se si se pueda definir los colores individuales por pixel, por ejemplo:
Leds.writePixel(0,0, pwm rojo, pwm azul, pwm verde) no se si me explico bien, un es decir poder definir el valor de intensidad de cada color en el pixel, sin escribir solo los definidos de WITHE_COLOR por ejemplo. Si es posible hacer esto te agredeceria me brindaras un ejemplo de como se hace ya que me serviria  mucho.

Te agradezco infinitamente tu orientacion, un saludo

michaelkamprath, 2021-01-07:
_Sorry, I am going to try to respond the best I can using Google Translate to read your question._

Much like my response to the other question you cite, you have the bit order of the 74HC595 reversed. The MSB is Q7, not Q0. Please read more closed [the bit layout documentation in the README](https://github.com/michaelkamprath/ShiftRegisterLEDMatrixLib#bit-layouts). In your 4x4 case, simply reverse the order of the bits, such that row 0 is Q0 and column 0 (the left most red) is Q7.

michaelkamprath, 2021-02-08:
Given the lack of follow up, I am closing this issue. Please feel free to re-open if you have any further questions.

closed 2021-02-08 by michaelkamprath