michaelkamprath/ShiftRegisterLEDMatrixLib #16: how do I customize it for an LED matrix instead of RGBLED matrix?

P1L3C, 2021-02-08:


michaelkamprath, 2021-02-08:
[Here is an example](https://github.com/michaelkamprath/ShiftRegisterLEDMatrixLib/blob/master/examples/other/monochrome-32x16/monochrome-32x16.ino) of using the "monochrome" driver rather than the RGB driver. Do you have question on how to use it beyond what is illustrated in the example?

P1L3C, 2021-02-08:
I'm using an Uno with two 74hc595 to control a matrix in this configuration

Q0 [U1] | 16, c8
-- | --
Q1 [U1] |  c7
Q2 [U1] |  c6
Q3 [U1] | c5
Q4 [U1] | c4
Q5 [U1] |  c3
Q6 [U1] |  c2
Q7 [U1] | c1
Q0 | R8
Q1 |  R7
Q2 | R6
Q3 |  R5
Q4 | R4
Q5 |  R3
Q6 |  R2
Q7 | R1

 I made a program for running text on an 8x8 matrix display. Now I want to make it so that the user could input their own text. I can't seem to find a solution... I know how to split a string into an array of characters, but I don't know how I can make an array of bytes to display on the matrix from the input chars... I thought this library would help make it

P1L3C, 2021-02-08:
Although I guess those drivers in the example aren't really needed... What's the pinout in the example? Is it the same as mine except that the rows are reversed?

michaelkamprath, 2021-02-08:
Your question is more about how to create an image to be displayed, more specifically how to collect and respond to a user's input. Realize this library is more about driving the LED matrix to show that image one that image has been made, and certainly does not enable or manage user input (something else has to do that). With that said, note that this library uses AdaFruit's GFX library as the API for creating the image to be displayed. This is a rich image manipulation API and does support printing text into the image. The example I cited above in fact uses the GFX library to draw text into the image that then this library in turn displays on the LED matrix.

If you are collecting a `char*` string from the user, the drawing that into the image is accomplished using the GFX library `print` method. 

[Here is a good tutorial on getting started with the AdaFruit GFX library](https://learn.adafruit.com/adafruit-gfx-graphics-library).

michaelkamprath, 2021-02-08:
Concerning your question whether your pin out is correct, I can't fully answer than without knowing how you wired the '595s to each other. For an 8x8 monochrome matrix, you need 16 bits to represent the rows and columns. The MSB bit should be column 0 (note I am using 0-based counting), and the LSB should be row 0, meaning column 7 is bit 7 (or bit 8 if you are doing 1-based counting), and row 7 is bit 8 (with 1-based counting, row 8 & bit 9). To accomplish this, the serial input from the MCU goes into the '595 that controls the rows, and the `QH'`output for that '595 feeds into the serial input of the '595 that controls the columns.  Given that wiring, column 0 is connected to `QH` on the column '595, and row 0 is connected to `QA` on the row '595.

P1L3C, 2021-02-08:
Ok, I connected everything right, but the text seems to be shifted down a few lines. I've adjusted the code for 8 rows and columns and adjusted the speed with loopcounter.

https://user-images.githubusercontent.com/78658446/107267228-e2c8d680-6a46-11eb-9fab-7a199c0815a5.mp4

michaelkamprath, 2021-02-08:
Note that my example code was meant for a 32x16 matrix, so simply changing the matrix dimensions in the code won't necessarily make things look right on a 8x8 matrix. Nonetheless, it does look like you have the hardware hooked up correctly and this driver library is correctly configured for your matrix design as the image I see in the video does seem reasonable for running the code as your described (e.g., the text font is larger than the 4 rows it is being printed to, hence you don't se the entire text pixels). Now you just have to create your image using the GFX API. Good luck!

P1L3C, 2021-02-08:
I fixed the issue with the shifted text, but can you help me get rid of the line at the top? I imagine that was put there as a kind of decorative frame. And yes, It's running correctly, thank you so much for that, I've been trying to make it work without a library, but this really helped!

P1L3C, 2021-02-08:
Okay, got rid of the line, but the text is now cropped at the top. Oh well, I think I can fix that by myself. Thank you for everything!

davidpilecek, 2021-02-08:
> Your question is more about how to create an image to be displayed, more specifically how to collect and respond to a user's input. Realize this library is more about driving the LED matrix to show [...]

I'm sorry, but I'm really confused about these libraries and functions. Is there a way for me to just modify the example above to not declare the str char. and upload its value with an input using serial.read() ?

michaelkamprath, 2021-02-08:
@davidpilecek Again, your question doesn't really pertain to this library specifically, but more around how do you program an Arduino in general. For the question you posted, [here is a tutorial on how to read strings from serial](https://www.instructables.com/Arduino-Function-Serialread-And-SerialreadString/). I am happy to answer questions, but please understand, the question you are presenting don't really pertain to this library.

davidpilecek, 2021-02-09:
I know how to read input from the user. The problem I have is the fact that the char str is used in the setup function, hence I don't know how I can read the input before the function setup of the library comes in.  Also, if I change the str in the getTextBounds function, It still doesn't work and prints out nonsense if I print it into the Serial monitor.
[image]

davidpilecek, 2021-02-09:
Or should I make a serial event?

michaelkamprath, 2021-02-09:
I am not really sure what you are getting tripped up on here. Have you tried rearranging the code such that you read the user input before setting up the library? Also, note that "setting up the library" and "drawing an image" are two different things, and thus can be implemented separately and in a desired order with respect to the collection of (user) data.

davidpilecek, 2021-02-09:
I tried putting the leds.setup function in the loop as well as getting the user input in the setup, which doesn't work. I'm not sure what else to do.

davidpilecek, 2021-02-09:
I slimmed down the code to this https://pastebin.com/kNNuJHZH (hopefully the delay won't be much of a problem), but it doesn't work... The display's showing two dots scrolling.

closed 2021-02-09 by davidpilecek