michaelkamprath/ShiftRegisterLEDMatrixLib #17: What is the useful life of an esp32 or esp8266 using the led matrix

alomdj00, 2021-03-08:
Hello, I know that calculating the useful life of something is very difficult, however I would like to ask you what is the maximum time you have used this library with usp32 or esp8266, I want to make my matrix to show text and animations all day practically, so I do not know if this library will force or quickly wear out the memory of an esp, and end up melting quickly, I know that many factors such as electrical and manufacturing influence, but let's say that the components and circuit are excellent, how long could this matrix last working, especially the esp.

michaelkamprath, 2021-03-08:
I use my `ShiftRegisterLEDMatrixLib` code in my own "LED art". I run it continuously. The only stability issues I have experienced has been with the power supply used, not the code I am running. 

Concerning "wear out the memory" on a ESP MCU, that concern only exists for writing to the flash memory, not the RAM. This code doesn't in itself write to flash.

alomdj00, 2021-03-08:
It's great, I was worried that the esp would melt in a couple days, I had that doubt because of what was between this super library and the ws2812, the latter being more expensive in my area, I will use a good power supply, of how many volts and amps you are using.
On the other hand, this library slows down to perform other tasks, takes up a lot of cpu time or I can use sensors normally.
You can lower the pwn levels to eliminate load to the cpu, I need to define the colors manually so I use the color definitions in the loop mediate serial or potentiometers, it is the same to show 256 pwm values ​​than to show 8 values ​​in terms of load or work From the esp, I don't really need millions of colors, with about 100 colors it would be more than enough for me.

Thanks a lot for everything, kind regards.

michaelkamprath, 2021-03-08:
Good luck with your project!

closed 2021-03-08 by michaelkamprath