michaelkamprath/ShiftRegisterLEDMatrixLib #18: [error: 'TCNT2' was not declared in this scope] When using a pro micro

hesa2020, 2021-05-05:
Hello I tried to use this library with my pro micro Leonardo-ish? (ATmega32U4) however I am getting errors the followings are not declared in scope:
TCNT2
TIMSK2
TOIE2
CS22
CS21
CS20
OCIE2A
ASSR
AS2
TCCR2B
WGM22
TCCR2A
WGM21
WGM20

I would really appreciate if you can apply a quick fix for this controller.
I purchased your kit thinking it would work nicely with a pro micro.

michaelkamprath, 2021-05-05:
The `ATmega32U4` is an older chip and doesn't have some of the support that more modern AVR Microcontrollers have. Specifically, it doesn't have the interrupt timer 2 that the code you are trying to use depends on for AVR MCUs, and generally has a different timer structure. My suggestion is that you use a more current MCU. The `ATmega1284P` is a good one to try. 

If I were to update the code here it would be to generate a compiler error when the `ATmega32U4` is being used.

michaelkamprath, 2021-05-05:
I see that SparkFun has created some new(ish) products using this `ATmega32U4` chip. I'll see what I can do but absolutely no promises on timeline or outcome. The `ATmega32U4` has a different timer set up than the `ATmega328P` (which the UNO is based on). This is an open source project, and pull requests are welcome. 

I'd strongly suggest upgrading you MCU, however. This library requires a good foot print of RAM due to its double-buffer technique and the 2.5K that the `ATmega32U4` has will only be sufficient for smaller matrices or single-color matrices. If you want to stick to AVR, the `ATmega1284P` works well. I'd recommend the Teensy 3.x or 4.x.

hesa2020, 2021-05-05:
@michaelkamprath Thanks I went to buy a Teensy 3.6 at a local store!

michaelkamprath, 2021-06-05:
@hesa2020  Did the Teensy work for you?

hesa2020, 2021-06-05:
I unfortunately never been able to get this working properly, im not entirely sure what im doing wrong

michaelkamprath, 2021-06-05:
But did you try with the Teensy? Also, how did you wire up the Teensy? Note that the Teensy has 3.3V output, and you will need to level shift the output to 5V. [I discuss that here](https://github.com/michaelkamprath/ShiftRegisterLEDMatrixLib#33v-logic-level)

michaelkamprath, 2021-08-07:
closing due to the lack of activity. reopen if new questions emerge.

closed 2021-08-07 by michaelkamprath