michaelkamprath/ShiftRegisterLEDMatrixLib #20: Monochrome 8 x 16 matrix

scrlk, 2021-08-09:
Hi, im trying to use your library to drive an 8 rows - 16 columns led matrix. Rows are turned on with a LOW signal and columns with HIGH, and columns are wired to the first three shift registers, rows are the last. Im using this constructor in your example, but it doesnt work (it just flickers randomly).
`LEDMatrix leds(8,24, HIGH, LOW, 1);`

Is there an extra setup step Im missing?
Thank you in advance

michaelkamprath, 2021-08-09:
The issue  is more likely hardware than software, either in how your matrix is wired or simply not using he right pins for on your microcontroller. Do you have a wiring diagram? What microcontroller are you using? Does the microcontroller output 3.3v and your matrix is built for 5v logic levels? Have you double checked the wiring?

scrlk, 2021-08-10:
I have double checkedmy wiring; I am using an atmega328p board with 5v logic. This is the wiring I am using, I have also tried switching the rows and columns order without success.
[image]

Im trying to drive my matrix with your library because of the adafruit gfx integration, I have checked the wiring of my matrix using the code below without libraries, but using your library would be much easier.

#define DATA 11
#define SHIFT 13
#define STORE 10

void store();
int col0[] = {0, 62, 8, 136, 136, 136, 8, 0};
int col1[] = {0, 0, 99, 20, 39, 64, 55, 0};
int col2[] = {0, 68, 78, 68, 68, 4, 76, 0};
int row[] = {1, 2, 4, 8, 16, 32, 64, 128};

void setup() {
  pinMode(DATA, OUTPUT);
  pinMode(SHIFT, OUTPUT);
  pinMode(STORE, OUTPUT);
}

void loop() {
  for (int i = 0; i < 8; i++) {
    shiftOut(DATA, SHIFT, MSBFIRST, ~row[i]);
    shiftOut(DATA, SHIFT, MSBFIRST, col2[i]);
    shiftOut(DATA, SHIFT, MSBFIRST, col1[i]);
    shiftOut(DATA, SHIFT, MSBFIRST, col0[i]);
    store();
  }
}

void store() {
  digitalWrite(STORE, HIGH);
  delayMicroseconds(10);
  digitalWrite(STORE, LOW);
}

michaelkamprath, 2021-08-10:
So if your own code generates an image, then I will say your matrix's bit layout is exactly opposite of the layout that this library expects. This library will shift out the column bit for the left most column first, works to the right most column, then the bottommost row, ending with the last bit being for the top most row. Your `loop()` above does not use that order. Basically, the shift register for the left most column is last (furthest form MCU) in the shift daisy chain, and the shift register containing the top most row is first (closest to MCU) in the daisy chain.

But fret not, it should be straightforward for you to alter this library to change the bit order expected for the shift registers. You want to look at `LEDMatrixBits::setRowControlBit()` and `LEDMatrixBits::setColumnControlBit()`, and if you are making a RGB matrix, you probably need to look at how the `_bitLayout` variable is used in the `RGBLEDMatrix` class, specify the bit ordering of the R, G, and B.

Remember, the first bit shift out goes to the highest line (`Q7`) on the furthest shift register.

scrlk, 2021-08-11:
Thank you for your help, I corrected my wiring accordingly, my order was wrong.

closed 2021-08-11 by scrlk

michaelkamprath, 2021-08-11:
So did you get an image on your matrix with this library? I'd love to see pictures of what you built!

scrlk, 2021-08-11:
I have not yet found the use of this display but here is an Game of Life implentation: 
[image]